Kinovarobotics/kinova-movo #5: Node `spawner` in movo.launch fails to spawn?

tttor, 2018-01-18:
Hi All:

In my case, the node `spawner` of `pkg="controller_manager"` launched in [`movo.launch`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_simulation/movo_gazebo/launch/movo.launch) always fails to spawn the controller. 

Any clue for why the spawner fails?
What if using a node `controller_manager` as below? Any drawback?

My workaround is:
```
    <node name="controller_spawner" pkg="controller_manager" type="controller_manager"
        args="  spawn
                movo/right_arm_controller
                movo/right_gripper_controller
                movo/left_arm_controller
                movo/left_gripper_controller
                movo/torso_controller
                movo/head_controller
                joint_state_controller"/>
```

Thank you,
tor

martine1406, 2018-01-18:
From my understanding, running a controller_manager node with argument <spawn> will achieve similar behaviour to using the spawner.  The killing process of the controllers could be different between the two options (I am not enough an expert to tell you if it is the case). I would saw to stick with your work around for now. I would not find the source of your issue. I'll let you know if I have more insight

martine1406, 2018-01-18:
PS: The spawner works fine on my side... spawner does not fail... maybe some known issue in ROS Indigo (there are quite a number of threads on spawners that fail when looking on the web!)

tttor, 2018-01-23:
This issue remains, and I guess, is related to error msgs in #7.
Some relevant discussion include: [this](https://answers.ros.org/question/198929/running-controller_manager-spawner-with-mybotrobot_description/) and [this](https://github.com/ros-controls/ros_control/issues/195).

What if using a node of type `controller_manager` instead of `spawner`, (If there is no catch for doing so) ?

martine1406, 2018-01-23:
I implemented your fix although the spawner did not seem to create any issue on my side.

tttor, 2018-01-24:
Thanks for the fix, and 
surely, we always look at whether there is any catch of  using `controller_manager` node.

closed 2018-01-24 by tttor