Kinovarobotics/kinova-movo #6: Set global_frame from map to odom in map_nav.launch?

tttor, 2018-01-18:
Hi All:

I launched [`demo.launch`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_simulation/movo_gazebo_demo/launch/demo.launch) and got:
```
[INFO] [WallTime: 1516239457.135306] [2.303000] Could not get transform from /map->base_link
[ERROR] [WallTime: 1516239457.135624] [2.303000] Could not get initial pose!!!! exiting....
```
My workaround is to set directly the `global_frame` from `map` to `odom` in [`map_nav.launch`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_demos/launch/nav/map_nav.launch).
Is it the right solution? or ?

Thank you,
tor

PS: this is using:
kinova-movo-master: Latest commit 8936d75
Indigo, trusty

martine1406, 2018-01-18:
Hi Vektor

I could reproduce your issue on some trials. I think it occurs when the map has not been loaded yet. Then, as soon as the map becomes available, the demo can proceed as planned. I would not change the global_frame. You may increase the time in demo.launch before launching movo_moveit_config movo_moveit_planning_execution.launch  and movo_gazebo_demo init_sim.xml if you want to give more time to the map to load. 

On my side, the demo is doing what it is supposed to do: i.e. go to home and open grippers, then go to tucked position and close grippers. 
Let me know if it is not the case on your side.  Thanks 
Martine

PatrickHussey, 2018-01-19:
You should use these launch files for the demos 
[simulation](https://github.com/Kinovarobotics/kinova-movo/tree/master/movo_demos/launch/sim)
[real robot](https://github.com/Kinovarobotics/kinova-movo/tree/master/movo_demos/launch/robot)

see instructions in [Readme](https://github.com/Kinovarobotics/kinova-movo/tree/master/movo_remote_pc_setup)

tttor, 2018-01-23:
I think this issue can be closed, 
as for mapless nav, we use [sensor_nav](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_demos/launch/nav/sensor_nav.launch), which sets `<param name="global_frame" value="/odom"/>`

closed 2018-01-23 by martine1406