Kinovarobotics/kinova-movo #7: sim_demo.launch is failing

tttor, 2018-01-23:
Launching [sim_demo.launch](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_demos/launch/sim/sim_demo.launch) as suggested for running simulation in https://github.com/Kinovarobotics/kinova-movo/issues/6, we got:
```
tor@movodev:~$ roslaunch movo_demos sim_demo.launch
...
[/home/tor/catkin-ws/src/kinova-movo/movo_demos/launch/demo/demo.launch] requires the 'is_sim' arg to be set
The traceback for the exception was written to the log file
[demo_bringup-6] process has died [pid 3176, exit code 1, cmd /home/tor/catkin-ws/src/kinova-movo/movo_common/si_utils/scripts/timed_roslaunch 35 movo_demos demo.launch sim:=true local:=true __name:=demo_bringup __log:=/home/tor/.ros/log/f8a3c91a-ffeb-11e7-9ad9-b8ca3ab91193/demo_bringup-6.log].
log file: /home/tor/.ros/log/f8a3c91a-ffeb-11e7-9ad9-b8ca3ab91193/demo_bringup-6*.log
...
[ERROR] [1516677519.458088619, 5.303000000]: "front_laser_link" passed to lookupTransform argument source_frame does not exist. 
[ERROR] [1516677520.585542905, 6.304000000]: "rear_laser_link" passed to lookupTransform argument source_frame does not exist.
...
[ERROR] [WallTime: 1516677620.982949] [100.993000] Timed out waiting for Gripper Command Action Server to connect. Start the action server before running example.
[WARN] [WallTime: 1516677621.140778] [101.138000] Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[init_sim_inplace-1] process has died [pid 4055, exit code 1, cmd /home/tor/catkin-ws/src/kinova-movo/movo_simulation/movo_gazebo/scripts/init_sim_inplace __name:=init_sim_inplace __log:=/home/tor/.ros/log/f8a3c91a-ffeb-11e7-9ad9-b8ca3ab91193/init_sim_inplace-1.log].
log file: /home/tor/.ros/log/f8a3c91a-ffeb-11e7-9ad9-b8ca3ab91193/init_sim_inplace-1*.log
...
```
And
```
tor@movodev:~$ roswtf
...
WARNING The following nodes are unexpectedly connected:
 * /robot_state_publisher->/init_sim_inplace (/tf)
 * /movo_marker_ctrl->/init_sim_inplace (/tf)
 * /movo_marker_ctrl->/rviz (/movo_marker_ctrl/update_full)
 * /move_group->/init_sim_inplace (/move_group/result)
 * /move_group->/init_sim_inplace (/move_group/feedback)

WARNING These nodes have died:
 * demo_bringup-6
 * init_sim_bringup-5
 * spawn_gazebo_model-4
 * default_controller_spawner-5
...
```
Using: https://github.com/Kinovarobotics/kinova-movo/commit/f7c370f81e140fe2aa6667ae2bcba4db2ebed117  on Indigo, trusty

martine1406, 2018-01-23:
implemented the suggested fixes

<include file="$(find movo_demos)/launch/nav/sensor_nav.launch">
    <arg name="local" value="$(arg local)"/>
    **<arg name="sim" value="$(arg sim)"/>**
</include>

and 

    <node name="demo" pkg="movo_demos" type="inplace_pick_place_demo.py" output="screen" >
    	<rosparam command="load" file="$(find movo_moveit_config)/config/kinematics.yaml" />
    	<param name="table_height" value="0.74"/>
    	<param name="sim" value="$(arg **sim**)"/>
    </node>

tttor, 2018-01-24:
Thanks for the fixes, it runs OK now.

closed 2018-01-24 by tttor