Kinovarobotics/kinova-movo #9: Error instantiating KinovaAPI object

aaronsnoswell, 2018-02-26:
Hi Kinova,

Following on from the discussion at https://github.com/orgs/Kinovarobotics/teams/movo-beta-program/discussions/15 I'm trying to implement a joint space velocity controller for our MOVO robot. In doing this I have been referencing the following files as a guide;

 * [kinova_api_wrapper.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py)
 * [jaco_joint_controller.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py)
 * [movo_jtas.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_jtas/movo_jtas.py)
 * [movo_arm_ctl](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/bin/movo_arm_ctl)
 * [movo_manipulation.launch](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_robot/movo_bringup/launch/manipulation/movo_manipulation.launch)

I have created a python ROS node that tries to instantiate a [`KinovaAPI`](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L154) object. Unfortunately, I get an error every time I try and construct this object.

```
movo@MOVO1:~$ rosrun popcorn_planning velocity_listener.py
[INFO] [WallTime: 1519619767.200056] Init API with arm=right, interface=eth0, robotIpAddress=10.66.171.15, localCmdport=25000, localBcastPort=25000, robotPort=55000
[ERROR] [WallTime: 1519619767.201531] Init API result:   1010
[ERROR] [WallTime: 1519619767.201661] GetDevices result: 2101
[ERROR] [WallTime: 1519619767.201748] Number of arms:    0
[ERROR] [WallTime: 1519619767.201828] Initialization failed, could not find Kinova devices                           (see Kinova.API.CommLayerUbuntu.h for details)
Segmentation fault (core dumped)
```

[image]

The error message refers to `see Kinova.API.CommLayerUbuntu.h` but this file is a header for a precompiled lib, so doesn't help me at all. It seems that the internal API construction (specifically [line 211](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L211)) is indicating that there are no arms attached.

 * Is the Kinova API object designed to be singleton? I.e. if there is already one of these objects, should I be able to create another one?
 * Our lab's development repo for the MOVO is private, but I have uploaded the source code for the relevant node to a [Gist that you should be able to see](https://gist.github.com/aaronsnoswell/002caa9c42c256c11fdea8324f7a7eca). Could you have a look and see if I am constructing the KinovaAPI object correctly? As you can see above, I'm using the paremters "arm=right, interface=eth0, robotIpAddress=10.66.171.15, localCmdport=25000, localBcastPort=25000, robotPort=55000".

Thank you for your assistance.

martine1406, 2018-02-26:
Hi @aaronsnoswell 

Yes, you should have only one instance of the Kinova API object. If you want to communicate directly with the arms using your listener, your you have to kill the regular movo nodes (i.e jaco_joint_controller.py) that talks to the arms. You cannot have jaco_joint_controller.py and velocity_listener.py talk to the arm at the same time. In this case, velocity_listener.py receives error 1010 because jaco_joint_controller.py is already talking to the arms.

Another option, since KinovaAPI is already created by SIArmController in jaco_joint_controller.py, would be to modify  jaco_joint_controller.py to listen to a topic published by your velocity publishing node and then pass on the message to KinovaAPI. Hence, the code in your velocity_listener.py could be moved and become part of the SIARMController "new version of yours".

You connection parameters "arm=right, interface=eth0, robotIpAddress=10.66.171.15, localCmdport=25000, localBcastPort=25000, robotPort=55000" are right and are equivalent to the configurations used in jaco_joint_controller.py

I hope this answers your question.
We're getting closer :-)
Martine

aaronsnoswell, 2018-02-26:
Thanks for confirming this Martine. I'll test again today and report back. Indeed - getting closer :)

aaronsnoswell, 2018-02-28:
Hi again Martine,

I'm still not having any luck. 

So far as I can tell, the only nodes that instantiate `KinovaAPI` objects are `/right_arm_ctl` and `/left_arm_ctl` through the dependency chain [movo_manipulation.launch](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_robot/movo_bringup/launch/manipulation/movo_manipulation.launch) > [movo_arm_ctl](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/bin/movo_arm_ctl) > [movo_jtas.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_jtas/movo_jtas.py) > [jaco_joint_controller.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py) > [kinova_api_wrapper.py](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py) (e.g. see [here](https://github.com/Kinovarobotics/kinova-movo/search?utf8=%E2%9C%93&q=SIArmController&type=) for confirmation).

I tried killing these two nodes with `rosnode kill <nodename>`, and still got the same error message as before. I then tried commenting out lines 9-32 in [movo_manipulation.launch](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_robot/movo_bringup/launch/manipulation/movo_manipulation.launch), and doing a rebuild and reboot to ensure these nodes are not present. I **still** get the same error message;

[image]

:(

 - Can you advise why I still can't communicate with the arms through the `KinovaAPI` object?
 - At this point, I am going to try and modify fork the `kinova-movo` repo and modify the contents of [SIArmController](https://github.com/Kinovarobotics/kinova-movo/blob/3272a4b33a46a1d63e86216f076436334fe3c7b0/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py#L51) to directly listen for our messages so we can do velocity control. I don't like this approach however, as it will require me to integrate upstream changes made to the `kinova-movo` repo. Do you know of any near-term plans to update MOVO to support joint velocity control through a ROS topic natively?

Thank you,

martine1406, 2018-02-28:
Hi @aaronsnoswell 

Comparing with the default config, I found that you set localCmdport to 25000 while we set it to 25000 by default, but you also set **localBcastPort to 25000, while we set it 25025**. Your local command port and local broadcast port cannot be the same. Sorry for not catching this earlier.... I must have read your initial post too quickly!

The errors you get indicate that the API is not initialized correctly. I tried killing /right_arm_ctl and /left_arm_ctl just like you, then running movo_arm_ctl (I had to change gripper = rospy.get_param("~gripper", ' ') to gripper = rospy.get_param("~gripper", 'kg3') in that file, but this change is uncessary when you launch from movo_manipulation.launch). If I am changing localBcastPort to 25000 just like you, I do get the same error. So I think we found the source of the error. 

Let me know
Martine

aaronsnoswell, 2018-03-01:
Thanks Martine - that config error turned out to be it. If I kill the other nodes and fix that port, I can construct the API object without error now.

closed 2018-03-01 by aaronsnoswell