Kinovarobotics/kinova-movo #15: Add support for velocity control modes

aaronsnoswell, 2018-03-14:
Hello Kinova,

At UQ we find ourselves in need of a way to command angular velocities on the MOVO robot arms. We are able to do this in some tests (e.g. see discussion at [this post](https://github.com/orgs/Kinovarobotics/teams/movo-beta-program/discussions/15)), however this requires killing the Kinova arm control nodes, which contain lots of useful safety-relevant code. Below I will outline a proposal to update this repository to natively support angular velocity control of the arms via a ROS interface.

I am happy to do the bulk of the work outlined below. I am seeking input from Kinova to steer the development work in an appropriate direction to get it merged as a pull request down the track, if appropriate.

aaronsnoswell, 2018-03-14:
# Modifications to kinova-movo to support velocity control

## Executive Summary
The kinova-movo codebase does not currently support angular velocity control of the JACO arms via any ROS interface. I am proposing to modify the codebase to add support for angular velocity control of the arms via a ROS interface. At a high level, the proposed modifications outlined here would update the kinova-movo codebase in the following ways;

 - Re-use the existing code for angular position control
 - Add new code to support angular velocity control 
 - Add a placeholder method (no implementation yet) for cartesian position control
 - Re-use the existing code for cartesian velocity control
 - Add necessary configuration methods to configure the robot for one of the above four control modes

## Detailed Summary
My understanding of the current MOVO arm control architecture is outlined in the below diagram;

[image]

The communication links from [`MoveGroupInterface`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_third_party/moveit_python/src/moveit_python/move_group_interface.py) to [`SIArmController`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py) all seem to be angular position trajectories i.e. a series of points, each point being a (vector of angles, timestamp) tuple (interestingly, [`SIArmController`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py) appears to support soft velocity and acceleration constraints for these trajectories, although this doesn’t seem to be exposed to any higher layers in the stack - e.g. see [line 77 in jaco_action_client.py](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_action_clients/jaco_action_client.py#L77) where the velocities are hard-coded to a zero vector).

[`SIArmController`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py) does one of two things based on the [`self._ctl_mode`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py#L147) member variable.
 - If [`self._ctl_mode`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py#L147) == [`AUTONOMOUS_CONTROL`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L157), it runs an internal software PID loop to track the most recent commanded trajectory, then outputs angular velocity commands to [`KinovaAPI`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py) via the [`KinovaAPI.send_angular_vel_cmds()`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L287) method
 - If [`self._ctl_mode`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py#L147) == [`TELEOP_CONTROL`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L158), it bypasses the PID loop and instead just directly commands cartesian velocities using the [`KinovaAPI.update_cartesian_vel_cmd()`](https://github.com/Kinovarobotics/kinova-movo/blob/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51/movo_common/movo_ros/src/movo_joint_interface/kinova_api_wrapper.py#L275)

I am proposing here that the MOVO codebase should expose an API to ROS users that covers each of the below control modes;
 - Angular Position Control (already implemented by Kinova as outlined above)
 - Angular Velocity Control (I am proposing to implement this by bypassing the PID loop in [`SIArmController`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py))
 - Cartesian Position Control (I am proposing to create placeholder methods for this mode, but no implementation – leave this work to Kinova in the future)
 - Cartesian Velocity Control (already implemented by Kinova, but only available via joystick teleop – I am proposing to add placeholder methods for exposing this via a ROS interface, but leave the implementation of this to Kinova in the future)

I plan to carry this work out in a branch at https://github.com/aaronsnoswell/kinova-movo based on https://github.com/Kinovarobotics/kinova-movo/commit/380c7827a9cec12cc0f4f0cc19ce0b88775c6c51 (latest commit at time of writing), and submit a pull request once the implementation is tested at my end. **As stated above, I am happy to do the bulk of this work, but I am seeking input from Kinova to steer the development work in an appropriate direction to get it merged as a pull request down the track, if appropriate.**

aaronsnoswell, 2018-03-14:
## List of proposed changes

**In kinova_api_wrapper.py**
 - [ ] Update the `AUTONOMOUS_CONTROL` and `TELEOP_CONTROL` flags
   - These flags actually mean angular velocity control and cartesian velocity control respectively. Rename these to actually mean what they say.

**In jaco_joint_controller.py**
 - [ ] Modify `SIArmController._ctl_mode`
   - Currently this is one of
      - `kinova_api_wrapper.py::AUTONOMOUS_CONTROL`, meaning angular position control with soft time, velocity and acceleration constraints, implemented via PID loop, then passing angular velocities through to `KinovaAPI`
      - `kinova_api_wrapper.py::TELEOP_CONTROL`, meaning simple cartesian velocity control – passes commands through to `KinovaAPI`
   - Adjust to be a 4-enum
      - `ANGULAR_POSITION`, same as `AUTONOMOUS_CONTROL` above. Uses PID loop to do angular position control with soft time, velocity and acceleration constraints, passing angular velocities through to `KinovaAPI`.
      - `ANGULAR_VELOCITY`, bypasses PID loop to directly drive velocities. **This is what we at UQ are actually interested in.**
      - `CARTESIAN_POSITION`, propose to mark as not implemented yet – leave to Kinova
      - `CARTESIAN_VELOCITY`, same as `TELEOP_CONTROL` above – directly passes cartesian velocities through
   - This will require updating the functions `_update_teleop_cmd`, `_run_ctl` and `__init__`
 - [ ] Add a class method to explicitly change the mode

## Discussion points

If the above changes are made, the SIArmController class would now support angular velocity commands, however there would not yet be a way to command this from ROS. I'm not sure of the best approach for this - I'd love Kinova's input. One idea I had is as follows;

 1. Add a topic (e.g. `/right_arm_controller/control_mode` or `/left_arm_controller/control_mode`) that is of some enum type (maybe just an int). This would specify which of the four proposed control modes the arm should be in.
 2. Update MovoArmJTAS to add a subscription to the above topic, and also to something like `/right_arm_controller/angular_velocity`
 3. If the control_mode topic is set to angular velocity, then the JTAS will ignore incoming cartesian position trajectory goals and just instead pass through angular velocities that it receives on the above topic to SIArmController. Otherwise, it would act as a regular JointTrajectoryActionServer.

The problem with this approach is that it changes the function of JointTrajectoryActionServer from just doing joint space trajectory tracking, to more general 'arm control' via trajectories, or velocities. Any comments on this suggestion or the proposal in general?

martine1406, 2018-03-14:
Dear @aaronsnoswell 

Thank you for your initiative. I believe your plan makes sense and I would be happy to act as your reference/guidance on the Kinova side :-) If you want, we could even organise a small conference call to discuss this matter (although I know we have quite a time difference between us)

about your discussion point, I would update SIArmController as you suggest. However, I am not too sure about the need to change MovoArmJTAS...?

If it were me, I would definitively add a subscriber to a topic of type /right_arm_controller/angular_velocity in SIArmController. Then you could use and update self._ctl_mode (I little bit the way it is done in _update_teleop_cmd here https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo_joint_interface/jaco_joint_controller.py#L268) 
Then, the SIArmController will just pass through angular velocities that it receives on the /right_arm_controller/angular_velocity or /left_arm_controller/angular_velocity topic to the API when self._ctl_mode is ANGULAR_VELOCITY.

I would personally have another new node publishing on  /right_arm_controller/angular_velocity and  /left_arm_controller/angular_velocity. I am not too sure why we would need to involve MovoArmJTAS here? I see this a little bit similar to the teleop node that directly deals with the SIArmController without involving the MovoArmJTAS? Let me know if you see this differently. 

Thanks!
Martine

aaronsnoswell, 2018-03-20:
Thanks @martine1406,

I concur with your suggestions - no need to modify MovoArmJTAS, just put the velocity topic subscriptions directly in jaco_joint_controller.py.

I have performed the updates described above and tested them today. Angular velocity control works and angular position control (pre-existing code) works. I am unable to get any motion in the arms when trying the cartesian velocity mode (pre-existing code), but this issue persists when I roll-back my branch to the existing v1.1.0 release codebase, so i believe this is a regression somewhere in the `kinova-movo` repo updates since 937025b6216fecc726a2ba6264423b040079e10f, and not a result of anything in my code. I raised a [separate issue](https://github.com/Kinovarobotics/kinova-movo/issues/23) to discuss this separately.

If we can figure out where the teleop / cartesian velocity regression has come from, and resolve this, I'll submit a pull request for the updated code. At that point I'll also write a summary of the changes I made in my branch for you and your colleagues to review.

Thank you,

aaronsnoswell, 2018-03-22:
After re-testing today, it appears the cartesian velocity code is working fine. I don't know what this issue was that was causing the problem earlier in the week. All that changed is that the robot has been re-booted several times this week, and the codebase re-compiled on board.

martine1406, 2018-03-22:
Hi @aaronsnoswell

We will have to perform more tests to check if we can replicate this issue and if so, what is the most probable root cause. If now, I cannot really tell. Thank you for letting us know.

closed 2018-07-04 by martine1406