Kinovarobotics/kinova-movo #20: Way to update packages on movo2, movo1 without removing skin

aaronsnoswell, 2018-03-19:
To install new packages on the MOVO PC's the current suggested solution (e.g. see wiki page [here](https://github.com/Kinovarobotics/kinova-movo/wiki/How-Tos#update-code-base-on-movo1-and-movo2-pcs) or discussion [here](https://github.com/orgs/Kinovarobotics/teams/movo-beta-program/discussions/22?from_comment=1#discussion-22-comment-1) is to connect movo1 and movo2 to an organisational network with internet access so that `sudo apt-get install <pkg>` can work.

For movo2 this is fairly straightforward as the HMI panel ethernet port can be used, however for movo1 this requires removing the skin. This is not desirable, especially as e.g. on our robot, the thread of the screws holding the skin are stripped, making it very hard to get the skin on/off.

For the 1.1.0 release, I found a way to do this without connecting the movo PC's to the internet (see [here](https://github.com/Kinovarobotics/kinova-movo/releases/tag/v1.1.0)). However this is a slow manual process and prone to error.

I'd like to propose that a better solution would be if instructions could be provided so that the aptitude repository for Ubuntu 14 could be hosted either on a movo development PC, or on a flash drive that could be connected to the HMI port (e.g. see instructions [here](https://askubuntu.com/a/1043/21299) on hosting aptitude repos on a flash drive).

Potential drawbacks of this approach;

 * It doesn't provide a way to run `rosdep update` commands. Does this matter?

I'd love to hear any other thoughts about a better way to update movo1, movo2 packages without removing the skin.

aaronsnoswell, 2018-03-22:
I had another thought here: I wonder if it is possible to bridge the internet connection from the dev PC to the MOVO PCs? I'm not sure what this would look like, but I found e.g. [this post](https://askubuntu.com/questions/590920/ubuntu-14-04-as-a-gateway-router-and-a-firewall) talking about something similar on Ubuntu.

aaronsnoswell, 2018-04-09:
Thanks @martine1406. At our end we have had some discussion and are going to try and set up a local apt repository on our movo dev machine. This requires about 15GB per architecture, per Ubuntu version apparently. Once this is done we should be able to add the local PPA to the movo1 and movo2 apt configurations, then just run `apt-get` on each of them without hassles. I'll let you know how we progress.

In the mean time, I've [developed a script](https://gist.github.com/aaronsnoswell/12f91060604d469df15ad6ad54e157fd) that allows for apt-get type package installations by downloading on movo dev then copying them across. I'll share it here as it might be useful to the community.

martine1406, 2018-04-10:
10-4. Thanks for the info. Very useful indeed