Kinovarobotics/kinova-movo #21: Kinect power cable rubbing internally

aaronsnoswell, 2018-03-20:
Hi Kinova,

I was having a look inside the top of our MOVO today (front 'head' skin panel removed) and noticed that the Kinect power cable has some serious mechanical wear. See below.

[image]

You can't see it in the above photo, but the Kinect cable is actually worn through to the point where the internal metal sheath is visible.

I believe this rubbing is happening when the MOVO torso linear actuator is used, but we've hardly used this actuator, so I'm not sure if it was exercised lots before shipping to us?

aaronsnoswell, 2018-03-20:
I did some more investigation and confirmed that the linear actuator is responsible. After raising the torso, sometimes when you lower it the Kinect cable gets pinched (see below photos). The culprit is the lip on the top of the linear actuator support.

[image]

[image]

We've fixed this on our MOVO using some electrical tape and the addition of two internal zip ties. I suggest this is fixed in future MOVO builds :)

[image]

martine1406, 2018-03-22:
Hi @aaronsnoswell

Thank you for reporting this problem. I confirm that we have a solution (protective box) that will be available for the next movos we ship. We checked, and it is possible to retrofit this box on current movo units that were manufactured without. I'll ask if we can get you one of these box along with instructions on how to install it. 

Thanks
Martine

aaronsnoswell, 2018-03-29:
Awesome. Looking forward to receiving it. Thanks.

brandonjdehart, 2018-03-29:
Sounds good to me! Thanks, Martine!

closed 2018-07-04 by martine1406