Kinovarobotics/kinova-movo #22: movo_config.bash: MOVO_HAS_KINOVA_ARM_NDOF overwritten on updates

aaronsnoswell, 2018-03-20:
When updating the kinova-movo code on our MOVO robot, the file [movo_config.bash](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_config/movo_config.bash) gets overridden. This lead to the environment variables `MOVO_HAS_KINOVA_ARM_7DOF` and `MOVO_HAS_KINOVA_ARM_6DOF` being reset, and caused us some confusion (arm controllers wouldn't work) until we figured out the issue and amended it.

I'm not sure what the solution here is - maybe this file should be .gitignored?

martine1406, 2018-03-22:
Hi @aaronsnoswell 

I agree with you that having to reset your configurations manually can be cumbersome. I am not sure though that the best solution is to gitignore the movo_config.bash file as it may be possible that this file gets updated at some point to add new configuration parameters for the movo. 

Maybe the solution could be to keep a copy of your local configuration somewhere and merge your config file with the new one you get when you update your movo with a new code version from the github?

I will discuss with the team what they think is the best solution.
I am sorry for this trouble. 
Martine

brandonjdehart, 2018-03-26:
The solution that I've come to for this issue is just to update the copy of movo_config.bash that lives on the remote PC and then have it copied over to the movo computers whenever I update their code. It would be great if this could be incorporated into a dynamic interface somewhere for those of us who may be switching arms in and out frequently, as doing a full catkin cycle on all three computers whenever a manipulator or arm is changed will get annoying pretty quickly.

martine1406, 2018-03-26:
Hi guys

I think as it was suggested here that the solution for now will be for you to keep a local copy of your config file and be sure to copy paste at least your right configs for the arms/arm type/grippers from your local copy to the new code you got from the github. You can even automatise this step on your end by making a script copying and pasting the right configs for you. 

Of course, having these parameters dynamically configurable could be great, but at the moment, it would require a complete rethinking of the code. Maybe again we could instead think of a code that would be called let's say activate_7dof_configuration and that would execute all the bash commands needed to stop movo2, update the configuration on all three computers (through ssh), then restart movo2. Something to keep in mind I guess if this becomes a serious issue on your end. 

Thanks for your valuable feedback as usual
Martine

closed 2018-07-04 by martine1406