Kinovarobotics/kinova-movo #27: ROS doesn't start if the arm bases aren't powered.

mattcorsaro1, 2018-04-12:
If the arm bases are unpingable during startup, ROS fails to launch. Ideally, setting `MOVO_HAS_KINOVA_ARMS` and related environment variables to false in [`movo_config.bash`](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_config/movo_config.bash#L58) would skip the startup ping.

martine1406, 2018-04-13:
Hi @mattcorsaro1 

Do you suggest to put MOVO_HAS_KINOVA_ARMS to false by default?

martine1406, 2018-04-13:
Hi @mattcorsaro1 

Okay I do get your question now: even if you put those env variables to false, the movo still does not starts up if both bases are not pingable. So what you are asking for is if we could add variables or change the behaviour of MOVO_HAS_KINOVA_ARMS and related environment variables to false in movo_config.bash to skip the startup ping if desired. I talked with the team and we believe it would be feasible quite easily. We'll work on it. Thanks for the feedback

brandonjdehart, 2019-01-17:
Any news on this issue?