Kinovarobotics/kinova-movo #28: WiFi network not launched if Movo e-stopped while powered on

mattcorsaro1, 2018-04-18:
If the e-stop button is pushed in when the Movo is powered up, the WiFi connection isn't launched. It would be useful to have access to the computers while e-stopped.

martine1406, 2018-04-18:
Hi Matt

When you boot the system up, it checks the e-stop button is not pressed (it is a safety check to insure the e-stop and other movo parts are in a no-fault state). If you boot up the system with the e-stop on, the battery will not send current in the movo upper parts (including the movo2 computer that deals with the wifi). As soon as you release the e-stop, you'll see that movo2 computer turns on (you can remove the upper back portion of movo skin, where the arm bases are located, and you'll see the light on the computer), and you should have wifi connection. Then you can press the e-stop again and movo2 computer will stay powered. 

For now, it is needed that you release the e-stop button before powering movo up.

mattcorsaro1, 2018-04-18:
Makes sense, thanks for the explanation.

closed 2018-04-18 by mattcorsaro1