Kinovarobotics/kinova-movo #29: Movo Gazebo /movo/cmd_vel linear movement rotates

belgiumkansas, 2018-04-25:
Using the /movo/cmd_vel topic to send linear velocities to the gazebo sim causes the simulated movo to rotate around its its axis rather than translating to a linear movement. sending an angular velocity along its z axis works properly but the linear x and y cause the movo to rotate and "fly" rather than move linearly.

martine1406, 2018-04-26:
Hi @belgiumkansas 

Thank you for contacting us. May I have more details on your publisher command. 
For example, I tested

rostopic pub -r 100 /movo/cmd_vel geometry_msgs/Twist "{linear: {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}"

and I had no problem. 
I tested with movo_demos sim_teleop.launch

Thank you
Martine

belgiumkansas, 2018-04-26:
rostopic pub -r 100 /movo/cmd_vel geometry_msgs/Twist "{linear: {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}" works okay but if I push the linear x velocity to .3 the movo starts "taking off". Also bringing the linear y velocity above .1 will cause movo to barrel roll. here is a video to illustrate the response https://youtu.be/hX1mtKjPoaQ

Currently our solution has been sending ros service calls to gazebo and setting it back on the ground (which does the job but is just a hack and a rather awkward one)

It didn't matter which launch file was used. It feels like there is an issue with the urdf model so I played around with it some to no avail (mostly adjusting inertial values and masses). Also gravity seems to have a rather weak response? I feel like those twist commands are just rotating the movo along the linear_actuator_link and gravity is just kind of "pulling" the movo in those directions.

I've used gazebo 7 and 9 hoping it was a version problem but to no avail. If there are any other details I can give to help you diagnose this let me know.

martine1406, 2018-04-27:
Hi @belgiumkansas

I tested on my end, and I do not have the same issue :-O. For me 
rostopic pub /movo/cmd_vel geometry_msgs/Twist "{linear: {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}" 
rostopic pub /movo/cmd_vel geometry_msgs/Twist "{linear: {x: 0.3, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}" 
or 
rostopic pub /movo/cmd_vel geometry_msgs/Twist "{linear: {x: 0.0, y: 0.1, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}" 
all work fine. It seems indeed that the gravity is not applied right on your model... I am using Gazebo 2.2.6. May it be related to your Gazebo version? Can you use version 6?

Here are my Physics properties:
[image]

Let me know
Martine

P.S. I found out that you do not need to specify a rate for the publisher to have the robot continue moving in Gazebo. Not related to your question, but it simplifies the call to the publisher not to specify a rate.

martine1406, 2018-07-04:
Hi @belgiumkansas 
Any updates on that situation? Were you able to set on Gazebo 6?
Any other movo research group had a 'flying Movo'  problem in Gazebo?

Thanks
Martine