Kinovarobotics/kinova-movo #30: Kinect 2 Framerate

stefie10, 2018-04-28:
When running on a remote machine, the Kinect 2 framerate is something like 1hz.  This is not even the point cloud, but the framerate I get when running rostopic hz on one of the image topics.  When I run the same command on the movo2 machine, I get 30hz, which is more like I expect.

We see an improvement when running on the wired network (something like 15 hz), but I expect the wireless framerate to be somewhat similar.  Has anyone else experienced this problem?  Are there suggested fixes?

cbames, 2018-04-28:
What are the wireless devices on the movo? We're upgrading anathema to WIFI
AD, in hopes that we'll have enough bandwidth to get 15hz wirelessly.

Barrett Ames

tttor, 2018-04-28:
This may be relevant: https://github.com/orgs/Kinovarobotics/teams/movo-beta-program/discussions/28

martine1406, 2018-04-29:
Indeed, the discussion pointed out by @tttor describes well the current situation on the movo I think. It also explains why we believe the framerate is so slow.