Kinovarobotics/kinova-movo #39: Blown Fuse

stefie10, 2018-05-30:
We blew another fuse.  We confirmed that it happened when the arm was impeded from moving to its target destination.  So probably some kind of over current or something.  Can you suggest a fix?  it's annoying to change the fuse every time.

Thanks!

ylamontagne, 2018-05-31:
We are truly sorry for the inconvenience and understand that it can be very annoying to change a fuse every time an arm does not behave like expected and hit something. We scheduled a session next week with our Movo setup to perform a full investigation during which we will try to reproduce your problem. We will let you know our conclusion and possible fix. 
Best Regards!
Yves

stefie10, 2018-05-31:
Thanks!  Actually I was somewhat excited because it was the first time we reproduced it.  If you like, we can try again to impede the arm to verify that's what causes it.  But I'm pretty sure from what I've observed.

Thanks and keep us informed!

Another related question - is there a better 'tuck' pose that doesn't cause the arms to rest on the floor?  If they remained upright then they might not hit the floor which is what happened this time.

And also is there a way to add the floor/ground plane to the Moveit planning scene?

stefie10, 2018-05-31:
Does Movo have an IMU?

martine1406, 2018-06-01:
Hi @stefie10 

About your questions:
Yes movo has an IMU in the base.

In their tucked position,the arms are not supposed to touch the floor!? Are we talking about the tucked position the arms take at the end of the boot-up dance?

Finally, for the ground plane thing, I know you can add virtual objects as well as Kinect output in Moveit planning scene. Would you want to add a virtual ground plane or the ground as perceived by the real movo? In all cases, I am sure you can find the kind of information (and a way more detailed one than what I could provide you) in the Moveit forums and Q&A pages. 

All the best :-)
Martine

pragathip, 2018-07-12:
Hi @stefie10 and @ylamontagne . We might be having the same issue. One of the arms stopped working when it collided with the other. How did you debug this issue and is there any documentation that you had access to for changing the fuse?

mattcorsaro1, 2018-07-13:
If you try to reboot the robot, ROS wouldn't start if it can't ping an arm. The reason for ROS failing to restart is logged in `/var/log/upstart/movo-core.log`. An easier way to tell if the fuse is blown is to check the blinking lights directly under the Ethernet cable on the arm control boxes in the back of the robot. If there are no lights under one of the cable connections, that arm's fuse is probably blown.

The fuses you'll need to replace are on the left front side of the robot. I think there are 4 fuse slots, though the left-most is empty on ours. I believe the left-most fuse is for the left arm, and the one to its right is for the right arm. I'm not sure what the right-most fuse is for.

vidara1281, 2018-07-13:
As mention here, @pragathip  [https://github.com/orgs/Kinovarobotics/teams/movo-beta-program/discussions/42](url). Can you give us a detail on what state was the robot?

Anything is always helpful to find a solution.

pragathip, 2018-07-13:
The right arm was impeded from moving to the target during teleoperation, just like another post described. First it was rigid and would not move. Then we turned the Movo off. When we powered it back on, the right arm stays flexible and does not harden up after a few seconds like the other arm. The Movo does not complete its boot-up process and the arms do not go into homing position. 

We confirmed that the right arm was not being pinged. We opened up front-top panel of Movo and found the fuses. The one corresponding to the right arm had blown. So it seems consistent with what the others have described. We have ordered replacements fuses and are awaiting their arrival.

pragathip, 2018-07-18:
We replaced the fuse and everything works.

goretkin, 2019-01-31:
Is there documentation on the electrical system? Which fuse connects to which arm?

I'm re-opening this issue, since it appears to have been inadvertently closed.

nsaif-kinova, 2019-01-31:
Please contact support@kinova.ca for any hardware related issue.
Thank you.