Kinovarobotics/kinova-movo #45: Gmapper Issues with Movo

andrewhou1, 2018-07-25:
Hey everyone,

I'm currently having some issues with getting gmapper to work correctly with the movo. I've been able to create a map of our lab space and launch a visualization of the map and the movo in it for RViz. The issue is that when I try to send the Movo to a new spot by setting a navigation pose, the local cost map gets populated with all of these fake "obstacles", and it makes the movo move in a random direction that isn't towards the goal pose. I've done some debugging, and it seems like the /move_base/EBandPlannerROS/global_plan contains invalid floating point values. When I echo out this topic, I can see that the path plan does indeed contain nan values. Does anyone have any thoughts as to why this might be happening? 

I'm not sure if this is directly related, but we also noticed that when we try to ping out front_laser, we get no responses. However, in RViz we can see the lase scans visually, and they seem right, so somehow Rviz is receiving those values.

martine1406, 2018-07-26:
Hi @andrewhou1 

I've never got such an issue. I would say that if you are not able to ping the front laser, it is not a very good sign. I would suggest searching on the forums related to the ROS navigation stack. This is not necessarily specific to movo... maybe there is something with the planner or it will create wrong plans if the lasers stop to provide data....

Are you giving movo an initial estimate of its position in the map?
Thanks
Martine

andrewhou1, 2018-07-31:
Hi everyone,

We ended up solving this issue by running the Gmapper node on Movo2 and then running RViz visualization on our normal computer (connected to Movo2). This probably worked because our computer can't ping the lasers but movo2 can

closed 2018-07-31 by andrewhou1