Kinovarobotics/kinova-movo #48: Movo not charging

jdchang1, 2018-08-16:
Hello,

Last week, we found that the charger of our Movo was broken, so we have replaced the power block with a new one since then. We are, however, not sure if this will solve our problem. We were wondering if you could direct us through some troubleshooting steps. Currently the robot is completely discharged and it is hard for us to gauge whether our solution is properly working.

Jonathan

martine1406, 2018-08-16:
Hi @jdchang1 

Someone from our electrical team will communicate with you as soon as possible to guide you better and probably ask you more questions... most likely that we'll want to know more information on that replacement power block you've ordered/used and its differences with the original one.... 

Until then, please turn movo off and keep it off. Please wait for my colleagues instructions before doing anything, even attempting to recharge movo with your replacement power block. Do not try to turn it on until you have evidence that movo could charge (when movo is charging, there are leds that turn on and that indicate the level of charge). 
Normally, the battery protection system will shut down the movo before the batteries get 100% discharged, but this complete discharged state is quite bad. If, for some reason the batteries got too discharged, they cannot be recharged. If you get another problem with the charger next time, please try to turn movo off before it gets completely discharged and contact us. 

Thank you
Martine

jdchang1, 2018-08-16:
Hello,

The power block that we ordered is exactly the same as the one that we were using before. We made sure to check that all of the specifications as well as the model were exactly the same. Furthermore, we do not believe that the battery is at the state of severe discharge. We looked at the battery's manual and it stated that a green light means that the battery is functioning properly.

Jonathan

martine1406, 2018-08-16:
Hi @jdchang1 

Okay good to know. Thanks for letting us know those info.  I'd still wait for my colleagues' recommendations. I expect you are writing on this forum because we see that movo is not charging properly? Or you want to confirm some info before attempting the re-charge with your new power block? In all cases, my colleagues should be able to provide you with more troubleshooting steps. If needed, we can organize a call in the next few days to make sure you get all the information you need and you are back on track with your movo

Martine

jdchang1, 2018-08-16:
Yes we are part of the Brown group.

vidara1281, 2018-08-16:
Hello @jdchang1 

I have a few questions.

First tell me, how did you determine that the charger was broken? Is it because MOVO was not charging with the original charger? Have you measured the output voltage of the charger? It should be around 28VDC.

When you say that you change the power block you mean the battery charger (i.e Meanwell)?  I would advise not to remplace it as the OEM manifacturer of the mobile platform might have modified the voltage and current output (Vo & Io). I don't think it will cause a problem but still if there's any issue with this part. It would be best to let us know. So that we can assess the situation. 

The only thing that could lead me to believe that it is not charging is that there is no contact from the charger bracket and the Anderson connector (contacts). Can you verify that the contacts are not inward from the Anderson connector of the bracket and the battery charger. Can you take a close picture of both part?

[image]

You said that you looked at the battery manual and green light. Are you referring to the MOVO LED Status? Or the embedded battery of the mobile platform? 

On the LED Status, is the led blinking green? What about the the state of charge? If MOVO was charging the Status should be blinking green and the SOC waving toward green solid (to a state of 100%)

MOVO LED Status
[image]

Lastly, what is the breaker position? It is on the ON position (closed - breaker switch to left) or OFF position (open - breaker switch at right).

Looking forward for your answers.

Vidara
Test Engineer

jdchang1, 2018-08-16:
Hi Vidara!
We determined our charger was broken by testing its output voltage and noticing there was none. The wires were accidentally pulled and we tried putting on new Anderson connectors and so on, but there was no output voltage from the charger as far as we could tell.
Yes, we got a new Meanwell with the exact same specifications. It is the exact same make and model as far as we can tell and its output is the same.
I will send pictures shortly.
We were referring to the embedded battery on the mobile platform's status.
The LED Status is not blinking green when we plug in our new charger.
Finally, I am not sure what breaker you are referring to. Where is this located?

Thank you for your quick replies! This is extremely helpful

jdchang1, 2018-08-17:
We located the breaker switch. We have not touched this switch since we flipped it when we got the Movo and it seems to be in the correct position.

This is an important issue for us to be able to proceed with our research; would it be possible to have a video chat or phone conversation now or sometime very soon? That would be incredibly helpful and would probably accelerate this troubleshooting process.

vidara1281, 2018-08-17:
I understand your situation and we are here to help. 

My email is vthanvilay@kinova.ca. I'll send you my contact. We can have quick talk on the phone if you want or we can set up a meeting tommorow with my colleague.

jdchang1, 2018-08-17:
I have just e-mailed you with my contact information!

Jonathan

jdchang1, 2018-08-17:
Hello,

The Movo has not charged overnight. What steps should we take now?

Jonathan

vidara1281, 2018-08-17:
Hello,

Sorry for the delay I have been caught up with meetings this morning.

I have been thinking about something. You guys used the new charger but have you setup the Vo and Io of the charger? These part a adjustable. You said that you measured 30V which is the max output voltage but there is also the current limit, if lowered ajusted it's not enough to charge the battery. Can you adjust the voltage to 28 V as it was from the original and verify the current adjustment? Looking closely on the potentiometer, it looks like it's on the maximum current (8A).

[image]

Can you send me the picture of the connector of MOVO and the charger. Here I took some picture as example. The wires should be well installed in the housing.
[image]
[image]

Can you also disconnect the power connector that is attached to the mobile base? If you look below MOVO you'll see this connector. Can you take a close picture of the connector to verify that the 2 wires (red and black) are well inserted in the housing as well. I wonder if theses wires are looses as well.

[image]

If both wired look well inserted, reconnected the connector to the mobile base, and then plug the charger to the power connector of MOVO.
[image]

Let me know if it is charging, The LED Status should be blinking. With the breaker switch on the left you should  see this.

[image]

Because your MOVO is low in battery, I don't know if you'll see something but let me know about your observations.

jdchang1, 2018-08-17:
Hello,

We have made sure that the connectors are functioning as expected. We have just adjusted the current and the voltage of the power block. When we plugged it in, we did not see any LED indication, but we confirmed that the outputted voltage was 28V and that we turned the knob for current to the maximum (8A). We also tried turning the breaker off and turning the MOVO on while it was plugged in but there was still no response.

Thank you
Jonathan

vidara1281, 2018-08-17:
If the battery is already low, MOVO won't turn on. Even if it does, the low state of charge of the battery reported will fault the mobile platform thus shutting down the system. You won't get any response. No need to power up MOVO. The problem is the low level battery.

It takes time for the battery to charge. It's not instant. One thing I have observed is that the charger gets very hot when charging the batteries. We have to wait and see.

Now that you have verified the power connector, I wonder if it's the internal Anderson directly to the mobile base that is not making contact.

Can you disconnect the power connector from the mobile base and plug the charger directly instead.  As seen on the picture below, be aware that the connector is inverted compare to the main power connector

[image]

Again, put the connector first before plugging the charger to the AC.

jdchang1, 2018-08-17:
Hello again,

We've followed your instruction and plugged the charger in directly in the correct orientation. However, the charger is not getting hot and the robot doesn't seem to be charging.

Do you have any more suggestions?

vidara1281, 2018-08-17:
Hello,

It takes time to have the battery charge. I would suggested to wait and see. Now that we are connected directly, Is the breaker switch at on (left)? Can you show me your setup and connectors? I haven't seen any pictures yet.

I am truly sorry about the inconvenience that issue is causing you. I will addressed it at the project committee and gather more info as soon as possible.

Until then, let me know if there is anything.

jdchang1, 2018-08-17:
Hi Vidara,

Here are pictures of our charger connector and the external power connector (on the movo's exterior, not the one on the mobile base because our plastic exterior is still on).

[image]
[image]

We can confirm that the charger has the correct power output, and that the contact with the Anderson poles seems to be perfect. Furthermore, the  The charger is still not getting hot even after 30 mins, nor is there any indication of charging whatsoever. Let us know if you have any ideas, would like more pictures, or anything at all.

Thank you for understanding the urgency of this matter and working with us to try to fix it. Please keep us posted on any updates or additional suggestions.

jdchang1, 2018-08-20:
Hi Martine!

Thanks once again for that quick reply. Unfortunately Jonathan isn't always in the lab, so posting updates on the GitHub issue is better in general because it lets all of us see it. I will post this information on the GitHub as well.
I have just run through all the cases and here are the results:

Case 1 
When we press the start button, MOVO will turn on for a few seconds and then shutdown. 
The Status LED doesn't turn on. The state of charge LED's will cycle and settle on low charge indication (last LED is yellow and no other LED's are lit)
Case 2
We do not see either of the status or state of charge (SOC) LED's turn on.
Case 3
We have set the Voltage and Current limits via the potentiometers to be 28V and max current (we believe it is 8A). The voltage reading from the charger is almost exactly 28V and not 28.5 or 28.6.
Yes, I can confirm it is the case that the ports are exactly the ones you've said.
Case 4
We are certain our charger outputs 28V (we measured it with a voltmeter). When we connect the charger to the charging port and attempt to measure the voltage between BIB TB3 pins 2 and 4, when the charger is plugged, we notice that the voltage is NOT 28V, but a rapidly fluctuating number that seems to be between 0.1 and 2.5 mV. 
Case 5
The resistance value is approximately 0.2 Ohms, but fluctuates between 0.1 and 0.3 Ohms. Given all the information we have, I personally feel that this explanation is the most likely.

I hope these responses were precise and thorough enough. If you need more information, please let us know. Furthermore, it might be helpful for us to get on a video call via Skype or Google Hangouts.

jdchang1, 2018-08-20:
I have just run through all the cases and here are the results:

Case 1 
When we press the start button, MOVO will turn on for a few seconds and then shutdown. 
The Status LED doesn't turn on. The state of charge LED's will cycle and settle on low charge indication (last LED is yellow and no other LED's are lit)
Case 2
We do not see either of the status or state of charge (SOC) LED's turn on.
Case 3
We have set the Voltage and Current limits via the potentiometers to be 28V and max current (we believe it is 8A). The voltage reading from the charger is almost exactly 28V and not 28.5 or 28.6.
Yes, I can confirm it is the case that the ports are exactly the ones you've said.
Case 4
We are certain our charger outputs 28V (we measured it with a voltmeter). When we connect the charger to the charging port and attempt to measure the voltage between BIB TB3 pins 2 and 4, when the charger is plugged, we notice that the voltage is NOT 28V, but a rapidly fluctuating number that seems to be between 0.1 and 2.5 mV. 
Case 5
The resistance value is approximately 0.2 Ohms, but fluctuates between 0.1 and 0.3 Ohms. Given all the information we have, I personally feel that this explanation is the most likely.

I hope these responses were precise and thorough enough. If you need more information, please let us know. Furthermore, it might be helpful for us to get on a video call via Skype or Google Hangouts.

closed 2018-09-05 by vidara1281