Kinovarobotics/kinova-movo #50: xacro deprecation warnings on `movo_common/movo_description/urdf/manipulation/jaco/jaco_7dof.urdf.xacro`

goretkin, 2018-09-13:
Due to lines like
`<property name="shoulder_pan_joint_rpy" value="0 ${M_PI} 0" />`

Full message:
```
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /data/movo_description/urdf/manipulation/jaco/jaco_7dof.urdf.xacro
included from: movo_description/urdf/movo.urdf.xacro

```

martine1406, 2018-09-13:
Hi @goretkin 

Thank you for your feedback. Is this something blocking you in your work? I understand it is good practice, but since it is a warning and not an error, is this something that creates a problem in usability for you?

Thank you
Martine

goretkin, 2018-09-13:
Hi @martine1406, no it's not a blocker. I just thought I'd make a note, especially since the fix should be straightforward.