Kinovarobotics/kinova-movo #52: Clarification on devel vs master

goretkin, 2018-10-12:
Hello, we are curious about the intention between the `devel`, `master`, and other branches of this repository. Could the maintainers of this repo clarify in the README or in this issue?

martine1406, 2018-10-15:
Hi @goretkin 

I am not the official maintainer of this repo, our software engineering team at Kinova is. However, what I can tell you is our V&V (verification and validation team) is responsible for testing the master branch so it is our <validated> branch. The devel branch was intended for more development work, later to be merged to the master branch. What I can tell you is that the devel branch is functional and we have used it on some of the latest movo units. However, we did not have time to do another official V&V phase. The next one will be in February 2019, and at that time, we will merge devel in master officially. 

I hope that clarifies things for you.
Martine

tlpmit, 2018-10-15:
> I am not the official maintainer of this repo, our software engineering team at Kinova is. However, what I can tell you is our V&V (verification and validation team) is responsible for testing the master branch so it is our branch. The devel branch was intended for more development work, later to be merged to the master branch. What I can tell you is that the devel branch is functional and we have used it on some of the latest movo units. However, we did not have time to do another official V&V phase. The next one will be in February 2019, and at that time, we will merge devel in master officially.

Note however that we were told that the calibration data for the individual Kinects for each Movo unit is supposed to be stored at:

kinova-movo/movo_common/movo_third_party/iai_kinect2/kinect2_bridge/data/

But, the master branch only has 3 serial numbers there.  However, the “devel” branch has many more files, including the calibration for our Movo.  

So, I think this suggests that “devel” should be the branch we should use in practice.

Tomas

alexvannobel, 2018-10-15:
Hi all, 
Martine summed up the situation very well, and I can add some more details regarding the maintaining of this repo:

**As of today,**
 - devel : ROS Indigo development branch. This contains the most recent version of the MOVO ROS code.
 - master : Release branch tested by V&V team. The most recent release is getting old and as Martine said, the next one will come only at the beginning of next year. It is better to use the devel branch as of right now as it is more up-to-date.

**In the upcoming months,**
MOVO is going to switch to ROS Kinetic. We will then have two development branches:
 - indigo-devel (actual devel)
 - kinetic-devel
Each devel branch will have its corresponding release branch (exact naming still TBD), tested by the V&V team : 
 - indigo-release
 - kinetic-release
The master branch will then point to the latest ROS distro release branch.
More information about the Kinetic release will be shared with you in the upcoming weeks!

Hope this sums up well.
Alex

closed 2019-01-08 by alexvannobel