Kinovarobotics/kinova-movo #53: catkin_make error

jdchang1, 2018-10-14:
Hello, I hope you are well. I was wondering if we could get any assistance with the error that we keep getting when we try to follow the manual setup instructions for movo. We have confirmed that PCL has been installed properly and we are currently stuck at this stage when we run catkin_make. Thank you!

Jonathan

[100%] Built target simple_grasping
Linking CXX executable /home/sagadre/movo_ws/devel/lib/simple_grasping/grasp_planner_node
Linking CXX executable /home/sagadre/movo_ws/devel/lib/simple_grasping/basic_grasping_perception
/home/sagadre/movo_ws/devel/lib/libsimple_grasping.so: undefined reference to `pcl::search::KdTree<pcl::PointXYZRGB>::KdTree(bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/sagadre/movo_ws/devel/lib/simple_grasping/grasp_planner_node] Error 1
make[1]: *** [kinova-movo/movo_common/movo_third_party/simple_grasping/CMakeFiles/grasp_planner_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/sagadre/movo_ws/devel/lib/libsimple_grasping.so: undefined reference to `pcl::search::KdTree<pcl::PointXYZRGB>::KdTree(bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/sagadre/movo_ws/devel/lib/simple_grasping/basic_grasping_perception] Error 1
make[1]: *** [kinova-movo/movo_common/movo_third_party/simple_grasping/CMakeFiles/basic_grasping_perception.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

alexvannobel, 2018-10-15:
Hi Jonathan, 

I'm guessing this error might be caused by a different version of libpcl than the one ROS Indigo requires. Can you send me the output of the following bash command?
```
sudo apt list --installed | grep pcl
```
This may also be an error caused by the installation of libpcl. Do you see any CMake warnings related to libpcl?

Thank you,
Alex

jdchang1, 2018-10-15:
Hello,
The output that I get is
`
libcommons-httpclient-java/trusty-updates,trusty-security,now 3.1-10.2ubuntu0.14.04.1 all [installed,automatic]
libhttpclient-java/trusty-updates,trusty-security,now 4.3.3-1ubuntu0.1 all [installed,automatic]
libpcl-1.7-all/trusty,now 1.7.1-3+trusty3 all [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-1.7-all-dev/trusty,now 1.7.1-3+trusty3 all [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-1.7-bin/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-1.7-doc/trusty,now 1.7.1-3+trusty3 all [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-all/trusty,now 1.7+trusty1 all [installed]
libpcl-apps-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-apps-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-common-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-common-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-features-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-features-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-filters-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-filters-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-geometry-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-io-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-io-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-kdtree-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-kdtree-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-keypoints-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-keypoints-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-octree-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-octree-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-outofcore-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-outofcore-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-people-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-people-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-recognition-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-recognition-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-registration-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-registration-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-sample-consensus-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-sample-consensus-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-search-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-search-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-segmentation-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-segmentation-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-surface-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-surface-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-tracking-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-tracking-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-visualization-1.7/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
libpcl-visualization-1.7-dev/trusty,now 1.7.1-3+trusty3 amd64 [installed,upgradable to: 1.7.2-1+trusty6]
ros-indigo-jsk-pcl-ros/now 1.2.4-0trusty-20180724-063101-0800 amd64 [installed,upgradable to: 1.2.4-0trusty-20181006-113855-0800]
ros-indigo-jsk-pcl-ros-utils/now 1.2.4-0trusty-20180724-055321-0800 amd64 [installed,upgradable to: 1.2.4-0trusty-20181006-111158-0800]
ros-indigo-pcl-conversions/trusty,now 0.2.1-0trusty-20180517-021808-0800 amd64 [installed,automatic]
ros-indigo-pcl-msgs/trusty,now 0.2.0-0trusty-20180517-021037-0800 amd64 [installed,automatic]
ros-indigo-pcl-ros/now 1.2.9-0trusty-20180715-182134-0800 amd64 [installed,upgradable to: 1.2.9-0trusty-20181006-074007-0800]
ros-indigo-perception-pcl/now 1.2.9-0trusty-20180715-190720-0800 amd64 [installed,upgradable to: 1.2.9-0trusty-20181006-103709-0800]
`

Also, I am not getting any CMake warnings at the moment.

Best,
Jonathan

closed 2018-11-20 by jdchang1