Kinovarobotics/kinova-movo #56: Using MoveIt End Effector Goal Pose

jdchang1, 2018-11-20:
Hello,

I am currently trying to control the right arm on the MOVO (6dof jaco arms) through moveit and I am running into a MoveIt error where there is no end effector to set the pose. I looked into the moveit API to find that this error is thrown when there is no end effector link for moveit to command. I was wondering if you knew of a solution to move the arm to a goal end effector pose. Thank you!

Jonathan

martine1406, 2018-11-20:
Hi @jdchang1 

Did you change anything (e.g. removed the end-effector)? How is your [movo_config](https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_config/movo_config.bash) file configured?

mattcorsaro1, 2018-11-20:
(I think we have 7dof arms)

martine1406, 2018-11-20:
Hi

We can try to reproduce your issue, but we would need to know what commands you are sending or what code example are you trying to run.