Kinovarobotics/kinova-movo #59: Cannot get message from /sim_initialized

patlican23, 2018-12-11:
Steps to Reproduce:

1.  `roslaunch movo_demos sim_demo_show_basic.launch`
2. `rosrun movo_demos demo_show_basic.py `

Result: 
```
Traceback (most recent call last):
  File "/home/iumay/movo_ws/src/kinova-movo/movo_demos/scripts/demo_show_basic.py", line 67, in <module>
    movo_base = MoveBaseActionClient(sim=True, frame="map")
  File "/home/iumay/movo_ws/src/kinova-movo/movo_common/movo_ros/src/movo_action_clients/move_base_action_client.py", line 58, in __init__
    self.last_pose = self._helpers.GetCurrentRobotPose(frame)
  File "/home/iumay/movo_ws/src/kinova-movo/movo_common/movo_ros/src/movo_action_clients/helpers.py", line 101, in GetCurrentRobotPose
    self.tfl.waitForTransform(frame, "base_link", rospy.Time().now, rospy.Duration(1.0))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf/listener.py", line 74, in waitForTransform
    can_transform, error_msg = self._buffer.can_transform(strip_leading_slash(target_frame), strip_leading_slash(source_frame), time, timeout, return_debug_tuple=True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 122, in can_transform
    not self.can_transform_core(target_frame, source_frame, time)[0] and
TypeError: time must have a to_sec method, e.g. rospy.Time or rospy.Duration
Exception AttributeError: "'MoveBaseActionClient' object has no attribute 'move_base_client'" in <bound method MoveBaseActionClient.__del__ of <movo_action_clients.move_base_action_client.MoveBaseActionClient object at 0x7f7816966450>> ignored

 ```

martine1406, 2018-12-11:
are you trying to launch sim_dmeo_show_basic, then run demo_show_basic? are you on ROS Indigo?

patlican23, 2018-12-12:
> are you trying to launch sim_dmeo_show_basic, then run demo_show_basic? are you on ROS Indigo?

I am on ROS Indigo and yes, I am trying to launch sim_demo_show_basic.launch and then running the demo_show_basic.py

alexvannobel, 2018-12-12:
Hi @patlican23 ,

The sim_demo_show_basic.launch file already calls the python script you are call in your second step.
First, on lines 28-31, it calls the demo_show_basic.launch file: https://github.com/Kinovarobotics/kinova-movo/blob/devel/movo_demos/launch/sim/sim_demo_show_basic.launch#L28-L31

This launch file in turn calls the demo_show_basic.py script :
https://github.com/Kinovarobotics/kinova-movo/blob/devel/movo_demos/launch/demo/demo_show_basic.launch#L15-L25

Do you have a problem with the ```roslaunch movo_demos sim_demo_show_basic.launch``` command? Why do you need to launch the python script a second time with the ```rosrun movo_demos demo_show_basic.py``` command? 

Best,

Alex

patlican23, 2018-12-12:
Hi, 

Yes, you are right. I have changed the following part in the sim_demo_show_basic.launch as following:
```
    <node pkg="si_utils" type="timed_roslaunch"
        args="35 movo_demos demo_show_basic.launch sim:=false local:=true"
        name="demo_bringup" output="screen"/>
```
In any case, I am getting the similar result.  If I change back to `sim:=true` in the sim_demo_show_basic.launch file and just launch the `sim_demo_show_basic.launch` file without running `movo_demos demo_show_basic.py`, I am getting the following;

Result:
```
Traceback (most recent call last):
  File "/home/iumay/movo_ws/src/kinova-movo/movo_demos/scripts/demo_show_basic.py", line 67, in <module>
    movo_base = MoveBaseActionClient(sim=sim, frame="odom")
  File "/home/iumay/movo_ws/src/kinova-movo/movo_common/movo_ros/src/movo_action_clients/move_base_action_client.py", line 58, in __init__
    self.last_pose = self._helpers.GetCurrentRobotPose(frame)
  File "/home/iumay/movo_ws/src/kinova-movo/movo_common/movo_ros/src/movo_action_clients/helpers.py", line 101, in GetCurrentRobotPose
    self.tfl.waitForTransform(frame, "base_link", rospy.Time().now, rospy.Duration(1.0))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf/listener.py", line 74, in waitForTransform
    can_transform, error_msg = self._buffer.can_transform(strip_leading_slash(target_frame), strip_leading_slash(source_frame), time, timeout, return_debug_tuple=True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 122, in can_transform
    not self.can_transform_core(target_frame, source_frame, time)[0] and
TypeError: time must have a to_sec method, e.g. rospy.Time or rospy.Duration
Exception AttributeError: "'MoveBaseActionClient' object has no attribute 'move_base_client'" in <bound method MoveBaseActionClient.__del__ of <movo_action_clients.move_base_action_client.MoveBaseActionClient object at 0x7ff7ac116e10>> ignored
[demo-3] process has died [pid 4211, exit code 1, cmd /home/iumay/movo_ws/src/kinova-movo/movo_demos/scripts/demo_show_basic.py __name:=demo __log:=/home/iumay/.ros/log/9e148e1c-fe41-11e8-b679-f496347775db/demo-3.log].
log file: /home/iumay/.ros/log/9e148e1c-fe41-11e8-b679-f496347775db/demo-3*.log
```

patlican23, 2018-12-14:
Hi Alex, 
    

- Robot seems to be initialize correctly according to your description.

- If  I launch  demo_show_basic.launch, I got
`[INFO] [WallTime: 1544756623.270727] Could not get transform from /odom->base_link
[ERROR] [WallTime: 1544756623.271772] Could not get initial pose!!!! exiting....
[INFO] [WallTime: 1544756623.272754] Stopping the robot...
`

- Then if I run move_base move_base
Result:
`
[ WARN] [1544730670.252423634]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_link does not exist.. canTransform returned after 0.100633 timeout was 0.1.
`
rostopic echo /movo/voice/text
WARNING: topic [/movo/voice/text] does not appear to be published yet

- If  I launched sim_demo_show_basic.launch while parameters in the demo_show_basic.py    
are changed to following;   
`movo_base = MoveBaseActionClient(sim=sim, frame="map")
  sim = rospy.get_param("~sim", True)`

an then run `rosrun robot_state_publisher robot_state_publisher `

Result:
```   
[ERROR] [1544756487.150394774, 147.081000000]: Lookup would require extrapolation into the future.  Requested time 146.036000000 but the latest data is at time 121.932000000, when looking up transform from frame [front_laser_link] to frame [base_link]

```
And,  then running `rosrun move_base move_base`

Result : `[ WARN] [1544732205.964491932, 416.059000000]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101 timeout was 0.1.`

alexvannobel, 2018-12-17:
Hi @patlican23 , sorry for the delay.

I don't think launching all those nodes separately is a good way to troubleshoot the demo, since some nodes require other nodes to be started beforehand (the robot state publisher, for example, is a core node) and you could create new errors that way. I would launch only the ```sim_demo_show_basic.launch``` and look at the log to see if some nodes failed initialization of crashed unexpectedly. 

Have you made local changes to the launch files you are referring to? Maybe a "sim:=false" was introduced somewhere and the ROS nodes start looking for the real robot instead of the simulated one. You should also make sure no node is already running before you launch the demo (except your ROS master). 

With an up-to-date devel branch, I cannot reproduce your error messages. If you are also up-to-date with this branch, the problem may be a configuration problem : 
 - You could look at your movo_config.bash file to see if every variable is correctly set.
 - You could also look at your movo_network_config.bash file to verify your computer has a correct simulation config. You can also verify this in a terminal with ```echo $ROS_MASTER_URI``` and ```echo $ROS_IP``` commands.

Hope this helps you,

Best, 

Alex

alexvannobel, 2019-01-08:
Hi @patlican23 ,

Did you resolve your issue?

patlican23, 2019-01-18:
Hi, 

Sorry for the late return. Yes, I solved the issue. One of my colleagues suggested to change following lines in the demo_show_basic.py file, then I was able to run it without any problem.

- tmp_left = rospy.wait_for_message("/movo/left_arm/joint_states", JointState)
- tmp_left = rospy.wait_for_message("/movo/left_arm/joint_states", JointState)  --and all other lines require joint_state 

to

- tmp_left = rospy.wait_for_message("/joint_states", JointState)
- tmp_left = rospy.wait_for_message("/joint_states", JointState) ...

Thanks

closed 2019-01-18 by patlican23