Kinovarobotics/kinova-movo #60: Problem with cartesian control

ljklonepiece, 2019-01-03:
hi there,

i have a question about cartesian control of the arm via joystick.
As stated in the manual, i activate cartesian control mode of left arm by pressing #12 on the joystick (lean direction to control motion in x-y and joystick rotation to control vertical motion in z)

i suppose while the arm is moved in cartesian mode, the orientation of the end-effortor should not change, however i observed that its orientation is keeping changing.

I took a video for your reference.
https://www.youtube.com/watch?v=8-3YkLoaXhM

in this video, when i move the arm in x direction, the orientation of the end effector is kept constant. but when i move the arm in z direction, the orientation cannot be kept. also when i move the arm in y direction, the orientation changes a bit.

wondering if there is some problem with the cartesian velocity controller? or it is supposed to behave like such?

note: the x,y,z direction mentioned above is with respect to the base_link frame

===============================================================

I also wrote a cartesian position controller myself based on two rostopics below
/movo/left_arm/cartesian_vel_cmd
/movo/right_arm/cartesian_vel_cmd

the package is available below
https://github.com/ljklonepiece/movo_cartesian_server

my cartesian controller encountered the same issue as in the case of joystick control

Any ideas to improve the controller are much appreciated!

sbrisebois, 2019-01-03:
Hi,

I think that your arm is configured in rotating frame instead of fixed frame.
You may change this by using the Development center (use a USB connection, because this feature is not working by ethernet in some base firmware version). 

This parameter is located in the Advance setting section.

Let me know if this fix your issue.
Sylvain

alexvannobel, 2019-01-03:
To add a bit to Sylvain's answer: 
To access the arms'Smart Controllers, you will have to take off MOVO's back panel. You can see in the picture below the two screws you will have to unscrew:
[image]
You also should unplug the Smart Controllers'Ethernet cables while you configure the arm by USB, then replug them after you're done. 
Best, 
Alex

ljklonepiece, 2019-01-04:
Thank you Sylvain and Alex for the proposed solution!

A few followup questions
0. what are some of the use cases where configuring the arm in rotating frame is preferred? 
1. can i configure the arm to fixed frame by changing the codes in movo1 (as I understand, the driver of the arms are launch in movo1 machine)?
2. could you point to me where the develop center software can be downloaded? is Linux supported for this software?

sbrisebois, 2019-01-04:
Hi ljklonepiece,

By default, Kinova robots are configured in Rotating frame. In Rotating frame, the hand/gripper
rotates automatically to follow the arm’s motion in the horizontal plane. This gives a more
human-like behavior. In Fixed frame, the hand’s orientation will not change unless explicitly
commanded. The rotating frame is more likely when you control the arm with the joystick and want to eat or manipulate stuff more easily.

You can set the frame mode by software (API command SetFrameType(int frameType)), but it has to be done only once, and as I said, it's necessary to perform it by USB unless the arm base firmware is version is 6.2.5 and higher. In MOVO, the base are connected by ethernet, so you will need to open the back panel... I firmly recommand to perform it via the DEV Center

The dev-center may be downloaded on our website : 
https://www.kinovarobotics.com/en/knowledge-hub/all-kinova-products
At the bottom of the page, in the software section, choose the KINOVA® SDK JACO² link.
The SDK run under windows ans linux, but I recommand running it under linux, because under windows 8 or higher, the USB driver need a special tweak to be installed (disabling driver signature).

Sincerely,
Sylvain

ljklonepiece, 2019-01-05:
> Hi ljklonepiece,
>
> By default, Kinova robots are configured in Rotating frame. In Rotating frame, the hand/gripper
> rotates automatically to follow the arm’s motion in the horizontal plane. This gives [...]

Thank you Sylvain for the clarificaiton!

I have tried the your proposed solution, but i am not able to set the reference frame to fixed frame.
I took two screenshoots for your reference

[image]
[image]

I am able to click on the "Fixed frame", but i found no way to save or to apply this setting (close and reopen the window, the setting is reverted back to "Rotating frame"). 
is there anything I may have missed? 

the version of kinova api and sdk are 
KinovaAPI-6.0.2-amd64.deb
SDK_JACO2_1.4.2_install64

alexvannobel, 2019-01-07:
Hi @ljklonepiece ,

It is true that this particular setting is a bit misleading. 
You will have to click on "Fixed frame", and then reboot the Smart Controller for the change to apply. 
To reboot the Smart Controller, you simply have to turn off and on again the physical switch on the Smart Controller box. It is also advised to close the Advanced Settings window as you shut down the arm.

When the arm will be online again, if you reopen the Advanced Settings window, you should see your setting applied (make sure you select your arm's serial number on the upper right corner of the window).

Hope this helps, 

Alex

ljklonepiece, 2019-01-08:
Thanks Alex!

The issue has been resolved!

closed 2019-01-08 by ljklonepiece