Kinovarobotics/kinova-movo #61: MOVO left arm does not have enough torque on shoulder lift joint

ljklonepiece, 2019-01-04:
Hi there,

recently, i am testing motion planning for both arms on movo and encountered the following problem.

the movo is supposed to move its left arm and right arm to a predefined joint configuration. the right arm moves along its planned path perfectly but the left arm suffered from executing the intended path, i recorded a video for your reference
https://www.youtube.com/watch?v=E0THA4qcHKA

in the video, movo first tried to move its left arm to a joint configuration which is a mirror configuration to the right arm. However, in the process of executing the planned path, the left arm got stuck. The warning signal from Moveit reports the error code "PATH_TOLERANCE_VIOLATED", which means the motion of the arm was deviated from the intended path by certain amount. 

From my observation, I think the problem lies on the shoulder_lift joint of the left arm. as you can see, when the left arm is almost in full extension configuration, the should_lift joint is shaking, indicating it could not exert enough torque to execute the motion. meanwhile other joints were moving as planned and the should lift joint was left behind, which resulted in the error code "PATH_TOLERANCE_VIOLATED".

I also noticed the shoulder lift joint is warmer than the other joints, which may confirm my guess that the shoulder lift joint is not function properly.

may I seek for your help to resolve this issue?

nsaif-kinova, 2019-01-04:
Hello,
This may be a hardware issue, please contact support@kinova.ca
Sincerely,

alexvannobel, 2019-01-08:
Hi,
I am closing the issue since it is handled by support@kinova.ca.
Feel free to reopen it if you find out it is software-related. 

Cheers,
Alex

closed 2019-01-08 by alexvannobel