Kinovarobotics/kinova-movo #62: Movo won't start after removing one arm

brandonjdehart, 2019-01-17:
We have a Movo with two 7-DOF arms and KG3 grippers. We've removed one of the arms to be used separately, and now the Movo won't launch all of its ROS components. We changed the movo_config file to make sure it knows it only has one (right) arm and one (right) KG3 gripper and ran make clean/make on both computers, but that doesn't seem to have helped.

Any input and/or solutions would be welcomed. Any else have this issue and solve it already?

mattcorsaro1, 2019-01-17:
It's been a while since we dealt with this issue, but I believe the problem was that the ROS startup script tried to ping both arms. The workaround Kinova suggested was to change the unused arm's IP address to the other arm's IP in the configuration file so both arms would have pingable IPs.

brandonjdehart, 2019-01-18:
Both of the controllers are still in the unit, so pinging both of them is not an issue. The problem comes in when it tries to run ROS nodes that want to talk to the left arm. Gripper action servers don't start which kills the robot_init launch script, the model still includes a left arm, etc.

I was hoping we wouldn't have to do a deep dive into the code to find everywhere the left arm is being mentioned/used without checking if it exists first.

nsaif-kinova, 2019-01-18:
Hello Brandonjdehart,
By default MOVO is intended to be used with two arms configuration. We have a branch to use Movo with a single arm, the code can be found here: https://github.com/Kinovarobotics/kinova-movo/tree/feature/single_arm_support

To update your robot, I suggest you first follow the instructions to set up your remote computer (an Ubuntu 14.04 machine): https://github.com/Kinovarobotics/kinova-movo/tree/feature/single_arm_support/movo_remote_pc_setup 

(you can follow steps 1 and 2: 1. Install ROS and 2. Setup Kinova-movo). 

Before Step 2, you can cd to your home directory. 
At step 2, before` cd ~/movo_ws/src/movo_remote_pc_setup`, you can `cd ~/movo_ws/src/` and checkout to branch feature/single_arm_support (git checkout feature/single_arm_support). 
You should then get a message telling you:
Branch feature/single_arm_support set up to track remote branch feature/single_arm_support from origin.
Switched to a new branch 'feature/single_arm_support'

After the code has been checkout to your computer, you must modify the config file to match with your actual configuration:
You have to edit the following file : 
`~/movo_ws/src/movo_common/movo_config/movo_config.bash`

In the file you have to change the following lines to match with your current MOVO Hardware : 

> export MOVO_HAS_RIGHT_ARM_6DOF=false
> export MOVO_HAS_RIGHT_ARM_7DOF=true
> export MOVO_HAS_LEFT_ARM_6DOF=false
> export MOVO_HAS_LEFT_ARM_7DOF=true

Save the file and then you can carry on with the rest of the remote pc setup instructions. 

When prompted:
Do you want to update the robot now? If not the code from movo2 will overwrite the src directory.... 
*PLEASE ANSWER NO UNLESS SPECIFICALLY INSTRUCTED BY KNOVA, or unless you are absolutely sure this is what you want to do. 
Install latest software on Movo now? [y/N]

you can answer YES. This will update the code on your movo platform with this new code. In case of problem, you can always revert to the original version of the code (found on the devel branch). Here are the instructions:

On your remote computer:

`cd ~/movo_ws/src/`
`git checkout devel`
`sync_robot` (insure you are connected to movo, ideally via an Ethernet cable)

Let us know if you have any trouble, and we'll try to fix the situation as quickly as possible.

brandonjdehart, 2019-01-18:
Since the single arm branch wasn't based on any of the other branches, there's no way for me to see what files changed and which didn't without doing a manual file comparison locally. Is there any way to get a list posted here (or a diff would be even better) that shows specifically which files needed to be changed?

We've got enough changes made locally that switching to an entirely different branch which is not able to merge nicely with any of the other branches makes this a painful process.