Kinovarobotics/kinova-movo #63: bringup procedure organization, move away from `timed_roslaunch`

kwhuang88228, 2019-01-31:
If I am understanding correctly, the intent of `movo_system_local.launch`  is to bring up the same nodes as `movo_system.launch`, but all the nodes spawn on the local machine.

Unfortunately, this is achieved by code duplication, and as a consequence, the files appear to be inconsistent. For example, `movo_moveit.launch` is called from `movo2.launch`, but this launch file doesn't get called ever from `movo_system_local.launch`.

If it is possible to move away from `timed_roslaunch`, it might be easier to organize the bringup launch files, because we could then place the `machine` tags in a logical place: in or around `movo1.launch` and `movo2.launch`, instead of peppered across the different files, requiring the use of the `local` flag.

Are these `timed_roslaunch`s necessary? Can we get rid of them?

goretkin, 2019-01-31:
(incorrectly posted from someone else's account)

If I am understanding correctly, the intent of `movo_system_local.launch`  is to bring up the same nodes as `movo_system.launch`, but all the nodes spawn on the local machine.

Unfortunately, this is achieved by code duplication, and as a consequence, the files appear to be inconsistent. For example, `movo_moveit.launch` is called from `movo2.launch`, but this launch file doesn't get called ever from `movo_system_local.launch`.

If it is possible to move away from `timed_roslaunch`, it might be easier to organize the bringup launch files, because we could then place the `machine` tags in a logical place: in or around `movo1.launch` and `movo2.launch`, instead of peppered across the different files, requiring the use of the `local` flag.

Are these `timed_roslaunch`s necessary? Can we get rid of them?

NaderMansouri, 2019-02-01:
Hi Gustavo,

`timed_roslaunch`’s purpose is to insert a small delay so that the launch files are executed in the right sequence in case certain modules need time to be initialized. Some processes requires that other processes be already launched. The way we found how to do that was with timed_roslaunch, though this might not be the only solution. 
		
Reorganizing the bringup launch files is a good idea. Please let us know how it goes.

Best,

Nader