Kinovarobotics/kinova-movo #66: Kinetic Migration

NaderMansouri, 2019-05-01:
## Description

Hi All, the Kinova team created a migration procedure for Movo. Here it is attached to this issue. If you have any questions, please publish an issue or directly contact support@kinova.ca
[Official migration of Indigo to Kinetic (1).pdf](https://github.com/Kinovarobotics/kinova-movo/files/3135737/Official.migration.of.Indigo.to.Kinetic.1.pdf)

ruminahui, 2019-07-03:
Hi, I am currently in the process of upgrading MOVO to Kinetic using this migration procedure. 
Something I noticed while running the "setup_movo_pc_migration" script after upgrade to Ubuntu 16, is that under the "Install ROS Kinetic" section of the script, the ros server key might be outdated (http://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/) This was causing some errors while installing ros kinetic packages.

pragathip, 2019-07-08:
Adding to the previous post, here are few other issues that we encountered.

1. **Connecting MOVO2 to the network**
While connecting to our department's network, the DNS server connection was not being set through the DNS information in the DHCP response of our department server. We had to manually add information to /etc/resolv.conf  to resolve the issue. At this point, we have a script that adds this information to the file at /etc/resolv.conf on startup. We are not sure what settings should be changed to have this information automatically detected and added by Movo's network hardware.

2. **Using MOVO2 as an internet bridge for MOVO1**
The network name is a parameter in the script, _setup_internet_on_movo_.
sudo iptables -t nat -A POSTROUTING -o **eth1** -j MASQUERADE
This information is also available in other threads that discuss connecting to Internet.

3. **Changes to "setup_movo_pc_migration" script**
Just like the previous post, we had to change the ROS server key from 421C365BD9FF1F717815A3895523BAEEB01FA116 to C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654.
Additionally, the commands for _libfreenect2_ installation are appropriate for Ubuntu 14, but not Ubuntu 16. This information is also available on the Github page for the Kinect driver, https://github.com/OpenKinect/libfreenect2.

### sudo dpkg -i debs/libusb*deb
sudo apt-get install libusb-1.0-0-dev

### sudo dpkg -i debs/libglfw3*deb; sudo apt-get install -f
sudo apt-get install libglfw3-dev