Kinovarobotics/kinova-movo #67: incorrect reference for gazebo camera sensor and plugin

belgiumkansas, 2019-06-20:
## Description

the urdf description for the movo contains a reference to a link that does not exist
## Version

kinetic-devel (but looks the same on master branch)

## Steps to reproduce

1. launch the gazebo simulator, something like movo_gazebo movo_test_zone.launch
2. try and find the camera topics

## Code example

line 88 in kinect_one_sensor.urdf.xacro

`<gazebo reference="${prefix}_ir_link">`

## Expected behavior
load the sensor and camera plugin

## Any other information
there are no other references to `${prefix}_ir_link`  in the robot description so gazebo will not load the sensor and camera plugin

replaced with:
`<gazebo reference="${prefix}_ir_frame">`
which is a valid reference and the camera plugin loads correctly.

simonferronforget, 2019-10-18:
Hello @belgiumkansas,
Thank you for sharing this issue.

I just want to make sure with you if the correct solution was instead to replace **line 88 in kinect_one_sensor.urdf.xacro** with:
`<gazebo reference="${prefix}_ir_optical_frame">`

Regards,
Simon

belgiumkansas, 2019-10-21:
that's probably the better solution. the ir_frame was the first valid reference that I saw in the urdf.