Kinovarobotics/kinova-movo #75: Unstable connection to MoVoWifi

joshuawisc, 2020-02-28:
## Description

I am trying to use a Linux laptop to connect to the Movo using MoVoWifi and ssh into MOVO2 but the connection is very unstable. The connection gets stuck after a few minutes of usage or when trying to copy a file.

## Version

Branch: kinetic-devel
Movo's Configuration: Double Arm, 7DOF-S, One gripper connected

## Steps to reproduce

Pinging the Movo from the Linux laptop or vice-versa continuously eventually causes some problem in the connection:
1.  Connect the Linux laptop to the MoVoWifi wireless network.
2.
   a. Running `$ ping 10.66.171.1` from the Linux laptop gives the following output:
[image]

   b. Pinging from the Movo to the Linux laptop also causes an issue after some time:
[image]

## Expected behavior

I am expecting the ping to go on continuously until I stop it.

## Any other information

Trying to copy a file or trying to ssh from two terminal sessions simultanously also causes an issue with the connection and causes the terminal to get stuck. The Linux laptop is still connected to MoVoWifi when the error occurs, so the issue is not the laptop disconnecting completely from MoVoWifi.

belgiumkansas, 2020-02-29:
do a traceroute to see where the packets are being routed through. There's 2 nucs in the movo, MOVO2 sits at the top (excellent wifi coverage) MOVO1 sits in the base (lots of EMF interference). So if your network is setup with the access point on MOVO1 that could be why your connection is so bad.

We decided use a separate router as the access point (a nice little usb powered one) so we rearranged the whole network for our purposes. Which gave us the bonus of having 2 radios in the upper torso giving us the benefit of setting up a gateway to our local subnet for internets and whatnot.

Also whats at 10.66.171.1? the 2 nucs are at 100, and 101, the arms are at 15 and 16, lidar at 8, 9 and the stanley platform at 5. I'm not running my movo or I would check but now but i'm curious.

joshuawisc, 2020-03-04:
Thanks for the reply! On our movo MOVO1 and MOVO2 are at 10.66.171.2 and 10.66.171.1 respectively. The packets are going directly to MOVO2:
[image]

Could you tell us what device you are using as a router?

belgiumkansas, 2020-03-04:
A GL-AR750. Its usb powered so that makes it easy to integrate with the platform. you could also try checking the RSSI in the wireless file to see if a drop happens when you start pinging out.
```bash
watch -n 1 cat /proc/net/wireless
```
or maybe an issue on your side? try a different laptop?