Kinovarobotics/kinova-movo #76: Movo quickly powers off after being turned on

ericrosenbrown, 2020-02-29:
Hello,

We are currently having an issue where after hitting hitting the power button on movo to turn it on, the movo boots up, but then dies after being on for about 30 seconds. The back button starts by beeping yellow (which indicates booting up), but then switches to red (indicating booting down), and then shuts off. When I connect a monitor directly to the movo, I see linux bootup, but the robot powers off shortly after. The movo is plugged in, and appears to have 75-100% battery. Is there any suggestions for how to figure out the issue? Thanks!

ericrosenbrown, 2020-02-29:
It seems like the NUC is not having issues, and specifically that the power from the base of the robot just cuts out shortly after booting. Is there any suggested debugging methods for figuring out how to resolve this power issue?

ericrosenbrown, 2020-02-29:
Where can we find a data sheet for the board specs for the base of the robot? we are trying to figure out if there is a blown fuse causing the power issue, but we can't find a data sheet for the board online.

belgiumkansas, 2020-02-29:
have you tried cycling the battery switch in the base of the unit and starting off battery power?

ericrosenbrown, 2020-03-01:
That solved our problem, thank you so much!!

closed 2020-03-01 by ericrosenbrown