Kinovarobotics/kinova-movo #77: Error while launching MOVO in Ubuntu 18

gogolprem, 2020-04-17:
I followed all the steps as mentioned and installed MOVO in Ubuntu 18 in ROS melodic version. However, while launching lets say - **roslaunch movo_demos sim_demo_show_basic.launch** it returns me this error. Gazebo opens, Rviz is also launched, but this error keeps on showing:

**terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  Could not find library corresponding to plugin range_sensor_filters/LaserScanFootprintFilter. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.**
[movo_laser_filter-12] process has died [pid 7680, exit code -6, cmd /opt/ros/melodic/lib/laser_filters/scan_to_scan_filter_chain scan:=/movo/scan_multi scan_filtered:=/movo/base_scan_filtered __name:=movo_laser_filter __log:=/home/gogol/.ros/log/e8ac75b2-80c1-11ea-bac6-b88198e13b65/movo_laser_filter-12.log].
log file: /home/gogol/.ros/log/e8ac75b2-80c1-11ea-bac6-b88198e13b65/movo_laser_filter-12*.log

Can someone please explain why this is happening and how to resolve this issue? Thank you!

belgiumkansas, 2020-04-20:
looks like you didn't build the `range_sensor_filters` package. Try re-building your workspace and checking for any errors in the compilation of that package.

gogolprem, 2020-04-20:
I do not remember seeing any compilation errors, but nevertheless, how would I re-build my workspace?

alexvannobel, 2020-04-21:
Hi @gogolprem ,

MOVO is not supported for ROS Melodic on Ubuntu 18.04.

By curiosity, did you update both NUCs with Ubuntu 18.04 and then proceeded to migrate the MOVO workspace code to ROS Melodic? 

When I did it for ROS Kinetic, I ran into many problems, the most common being a third party package not being updated / supported for the new ROS version. Your error seems related to `range_sensor_filters` not functioning properly on ROS Melodic. As @belgiumkansas pointed out, you should first make sure this package works first. 

We don't actually support and don't plan on supporting MOVO on ROS Melodic, so I'm afraid I cannot provide many technical insights and solutions to your problem.

Sorry for the inconvenience,
Best regards,
Alex

closed 2020-04-21 by alexvannobel