Kinovarobotics/kinova-movo #85: Charger replacement?

zkytony, 2022-09-26:
MOVO accidentally stepped onto the charger while navigating and broke the green cable connecting to the black slot on the power connector

[image]

We could attempt a short-term fix to reconnect the cable and the power connector ourselves. I wonder if it's still possible to buy the charger somewhere? I can find the HLG-240H-30A power supply online, but it doesn't seem to come with the 8-pin power connector that MOVO needs. 

Also, any information regarding manually fixing the power cable connection is appreciated!

Thank you!

alexvannobel, 2022-09-26:
Hi @zkytony ,

As the Github is for software issues, you are better writing to support@kinova.ca to have information on the charger.

Cheers,
Alex

zkytony, 2022-10-05:
Hi, I tried emailing support through that email last week, but haven’t got an response. I also looked at the Contact Support page here https://www.kinovarobotics.com/reach-us/customer-support
And I selected “Gen2 robots”. But it requires a serial number. How could I know the serial number of MOVO?

Thank you!

felixmaisonneuve, 2022-10-17:
Sorry for the late reply.

Did you get any answer from the support team?
I think you could use the serial number of one or both the jaco2 arms.

felixmaisonneuve, 2022-10-18:
You can just write "N/A" as your serial number and the support team wil be able to find the serial number of your MOVO/Jaco arms only with your client informations (city, university name, company name...)

zkytony, 2023-03-10:
Update: we reached out to Kinova and ordered a replacement. We followed @felixmaisonneuve's suggestion when submitting the initial support ticket. Then, someone got back to us 2 days later, and we provided them the serial number of our robot. Then, the support was able to find a spare PSU unit and ship that to us.

closed 2023-03-10 by zkytony