Kinovarobotics/kinova-movo #95: movo-core.services not found, action services for left arm not started

amansingh427, 2023-03-15:
Hello all, we are having trouble with starting the required services for our MOVO after removing the right arm. Our MOVO has one left arm with 7 DOF and as seen below, we are trying to use the single arm configuration branch. Our MOVO is also using ROS Kinetic. 
 
We initially started by following the instructions posted in [this post](https://github.com/Kinovarobotics/kinova-movo/issues/62). Then we ran `catkin_make` in `movo_ws`. We made some small edits to make it compatible with Kinetic, and edited the movo_config.bash to change the following variables: 
```
MOVO_HAS_TWO_KINOVA_ARMS = false 
MOVO_HAS_RIGHT_ARM_7DOF = false 
MOVO_HAS_RIGHT_KG3_GRIPPER = false 
```
Whenever we try to start the ROS calls via `movostart`, we get the following error: `Failed to start movo-core.service: Unit movo-core.service not found`. We usually have to start the robot using `movo-core-start`, and that starts the services for all but the left arm (and left gripper). The MOVO is able to ping the left arm, but the action server for the left arm seems to never start; first a warning pops up, saying `Waiting for movo/left_arm_controller/follow_join_trajectory to come up`. The action client then times out due to not being able to connect to the action server. This only happens for the arm, as the body and the camera works just fine.

We think that these two issues might be related to each other, but we are not sure if they are, or how they would be related.
  
Any help would be greatly appreciated.

brandonjdehart, 2023-03-16:
When we were initially working with our Movo several years ago, we had a similar issue when we tried to only use a single, left arm. I don't believe we ever solved the issue for the left arm, but for whatever reason just switching the Movo to one use a single, right arm seemed to work fine. Our assumption at the time was that the code was expecting to connect to either no arms, one right arm, or two arms and a single left arm was never an expected operating mode.

KinovaCustomerSupport, 2023-03-17:
> ***
> Case: CS009061, [Kinovarobotics/kinova-movo] movo-core.services not found, action services for left arm not started (Issue #95)
> ***

/* RTE style begin */.rich ol, .rich ul { margin-top: 0px; margin-bottom: 0px;}
.rich .imgbrdr {
	border: solid 1px;
}
.richIcon {
	display: inline-block;
	background-image: url(Images/sprite_main5.png);
	background-repeat: no-repeat;
	width: 15px;
}
.richAttach {
	background-position: -2px -122px;
}
.richImg {
	background-position: -2px -1120px;
}
.richScreen {
	background-position: -24px -1032px;
}
.richArticle {
	background-position: -24px -1054px;
}

/* Correct box */
.rich .GrayBox {
	border: solid 1px #CCCCCC;
	background-color: #F9F9F9;
	padding: 4px;
	/*margin: 4px;*/
	width: 98%;
}
.rich .hintcell
{
	height: 36px;
	width: 32px;
	vertical-align: middle;
}

.rich .warncell
{
	height: 36px;
	width: 32px;
	vertical-align: middle;
}

.rich .boxcontent{ width: 100% }
.rich pre { font-size:1.3em }

.rich iframe{
	border: none;
}
.rich iframe.iframeBorder, .rich img.iframeBorder {
	border: solid black 1px;
}

.rtetable > tbody > tr > td, .rtetable > tr > td { white-space: nowrap }
.rtetable > tbody > tr:first-child > td, .rtetable > tr:first-child > td { background-color: silver }

.rich pre.source-highlighted {background-color: #F0F0FF;}
.rich pre.source-highlighted .nu, .rich .sourcetable .nu { color: brown; }
.rich pre.source-highlighted .kw, .rich .sourcetable .kw { color: blue; }
.rich pre.source-highlighted .cmt, .rich .sourcetable .cmt { color: green; }
.rich pre.source-highlighted .sl, .rich .sourcetable .sl { color: brown; }
.rich pre.source-highlighted .br, .rich .sourcetable .br {color: darkgreen; }

.rich pre.source-highlighted .add, .rich .sourcetable .add { background: palegreen; }
.rich pre.source-highlighted .del, .rich .sourcetable .del { background: pink; }

.rich table, .rich tr, .rich td, .rich img, .rich iframe {
	cursor: default;
}
.layout .rtecell {
	vertical-align: top;
	font-size: small;
}
.rtetable > tbody > tr > td, .rtetable > tr > td { vertical-align: top; }

.rich p{
	margin:0px 5px 0px 0px;
}
.rich .align-center{
	text-align: center;
}
.rich .align-right{
	text-align: right;
}
.rich .align-left{
	text-align: left;
}
.rich li[style*='text-align: center'], .rich li[style*='text-align: right'] {
list-style-position: inside;
}

.rtetable { empty-cells: show; border: solid 1px black; border-collapse:collapse; }
.rtetable > tbody > tr > td, .rtetable > tr > td { border: solid 1px black; height: 1.5em; min-width: 20px;  padding: 0px 5px }
.rich table {font-family: arial; font-size: small}
.rich blockquote.shift {
	margin-right: 0px;margin-top: 0px; margin-bottom: 0px;
}
.rich img[data-expand] {
	display: block;
}
.rich p, .rich p.richp  {font-family: arial; font-size: small}
.rich blockquote.shift h1, .rich blockquote.shift h2, .rich blockquote.shift h3, .rich blockquote.shift h4, .rich blockquote.shift h5, .rich blockquote.shift h6 {
    margin-top: 0px;
}
.rich, .rich * {
    word-wrap: break-word;
}

.sourcetable {background-color: #F0F0FF; font-family: monospace; width: 100%}
.sourcetable tr {
	margin: 0px;
	padding: 0px;
}
.sourcetable td {
	margin: 0px;
	padding: 0px;
	white-space: nowrap;
}
.sourcetable > tbody > tr > td:first-child, .sourcetable > tr > td:first-child
{
	width:1px;
}

.filelist {
	/*display:inline-block;*/ padding: 5px;
}

.rich .toc {
	/*display:inline-block;*/ padding: 5px;
}
.rich .toc .header,.rich  .wikilist .header,.rich  .filelist .header {
	font-size: 1.7em;
}
.rich .toc li,.rich  .filelist li {
	margin: 5px 0px;
}
.rich .wikilist div {
	margin: 10px;
}

.rich .layout {min-height: 2.5em; display:table; width:100%; border-spacing:5px; table-layout:fixed; empty-cells: show;}
.rich .layout .rtecell {display:table-cell; height: 2.5em}
.rich .layout .rteside {width: 25%}

/*
.rtetable {
	border-collapse:collapse;
}
.rtetable > tbody > tr > td, .rtetable > tr > td {
	border: 1px solid #ddd;
	padding: 7px 10px;
	vertical-align: top;
	text-align: left;
}*/
.rich blockquote.realbq {
	border-left-color: lightblue;
	border-left-width: 1px;
	border-left-style: solid;
	margin-top: 0px;
	margin-bottom: 0px;
	padding-left: 5px;
}
.rich body {font-family: arial; font-size: small}

.rich .GrayBox {
	margin-left: 8px;
}

.rich .sourcetable { font-size:1.3em;}

/* RTE style end */Hello Aman,Like Brandon mentioned, if you wish to use only 1 arm on the MOVO, you'll have to use the right one. You can then download the right branch to make it work: https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_config/movo_config.bash#L57-L64

amansingh427, 2023-04-11:
Thank you for your response. We moved the arm to the right, but now we are getting the same issue, with the action server for the right arm not starting. We are additionally getting this error when running `sudo movo-core-start`:
```
remote[movo1-0]: [ERROR] [1681258881.039189]: Init API result:   1
[ERROR] [1681258881.039554]: GetDevices result: 1
[ERROR] [1681258881.039764]: Number of arms:    0
[ERROR] [1681258881.039945]: Initialization failed, could not find Kinova devices                           (see Kinova.API.CommLayerUbuntu.h for details)
[right_arm_ctl-1] process has died [pid 4312, exit code -11, cmd /home/movo/movo_ws/src/movo_common/movo_ros/bin/movo_arm_ctl __name:=right_arm_ctl __log:=/home/movo/.ros/log/37b164fe-d8c8-11ed-998b-94c6911bfb50/right_arm_ctl-1.log].
log file: /home/movo/.ros/log/37b164fe-d8c8-11ed-998b-94c6911bfb50/right_arm_ctl-1*.log
```
We are able to ping both arms now (we fixed an issue where both arms had IP addresses 10.66.171.16), and the right arm seems to work by itself when connected to another machine via USB.

I look forward to your reply.

amansingh427, 2023-06-05:
Solved! While going through the arm initialization code, I found that the MOVO only detects an arm if the arm is at a specific port; for the right arm, that port is 55000. Using the Jaco SDK, I saw that the port was incorrect, so I changed it, and the arm now works as normal along with the MOVO. This is what my right arm looks like:
[image]

closed 2023-06-05 by amansingh427